jianboqi/CSF #2: CSF Python install on Windows-64bit

alexguyot, 2017-03-14:
Hello,

I'm using laspy on WinPython-64bit-2.7 + MinGW and would like to install the CSF python module, but the install is failing during build (see below).

I'm not familiar with building and installing from sources so if someone can point me to the right direction it would be great !

Thanks for your help.
Alex

---------------------------
running build
running build_py
running build_ext
building '_CSF' extension
C:\MinGW\bin\gcc.exe -mdll -O -Wall -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\
python-2.7.12.amd64\include -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\
..\python-2.7.12.amd64\PC -c csf_wrap.cxx -o build\temp.win-amd64-2.7\Release\cs
f_wrap.o
C:\MinGW\bin\gcc.exe -mdll -O -Wall -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\
python-2.7.12.amd64\include -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\
..\python-2.7.12.amd64\PC -c ../CSFDLL/c2cdist.cpp -o build\temp.win-amd64-2.7\R
elease\..\csfdll\c2cdist.o
In file included from ../CSFDLL/Cloth.h:46:0,
                 from ../CSFDLL/c2cdist.h:5,
                 from ../CSFDLL/c2cdist.cpp:1:
../CSFDLL/Particle.h: In constructor 'Particle::Particle(Vec3, double)':
../CSFDLL/Particle.h:31:7: warning: 'Particle::old_pos' will be initialized afte
r [-Wreorder]
  Vec3 old_pos; // the position of the particle in the previous time step, used
as part of the verlet numerical integration scheme
       ^
../CSFDLL/Particle.h:25:7: warning:   'Vec3 Particle::acceleration' [-Wreorder]
  Vec3 acceleration; // a vector representing the current acceleration of the pa
rticle
       ^
../CSFDLL/Particle.h:49:2: warning:   when initialized here [-Wreorder]
  Particle(Vec3 pos, double time_step) : pos(pos), old_pos(pos), acceleration(Ve
c3(0, 0, 0)), mass(1), movable(true), accumulated_normal(Vec3(0, 0, 0)), time_st
ep2(time_step)
  ^
../CSFDLL/Particle.h:25:7: warning: 'Particle::acceleration' will be initialized
 after [-Wreorder]
  Vec3 acceleration; // a vector representing the current acceleration of the pa
rticle
       ^
../CSFDLL/Particle.h:23:9: warning:   'double Particle::mass' [-Wreorder]
  double mass; // the mass of the particle (is always 1 in this example)
         ^
../CSFDLL/Particle.h:49:2: warning:   when initialized here [-Wreorder]
  Particle(Vec3 pos, double time_step) : pos(pos), old_pos(pos), acceleration(Ve
c3(0, 0, 0)), mass(1), movable(true), accumulated_normal(Vec3(0, 0, 0)), time_st
ep2(time_step)
  ^
../CSFDLL/Particle.h:23:9: warning: 'Particle::mass' will be initialized after [
-Wreorder]
  double mass; // the mass of the particle (is always 1 in this example)
         ^
../CSFDLL/Particle.h:21:7: warning:   'bool Particle::movable' [-Wreorder]
  bool movable; // can the particle move or not ? used to pin parts of the cloth

       ^
../CSFDLL/Particle.h:49:2: warning:   when initialized here [-Wreorder]
  Particle(Vec3 pos, double time_step) : pos(pos), old_pos(pos), acceleration(Ve
c3(0, 0, 0)), mass(1), movable(true), accumulated_normal(Vec3(0, 0, 0)), time_st
ep2(time_step)
  ^
../CSFDLL/c2cdist.cpp: In member function 'void c2cdist::calCloud2CloudDist(Clot
h&, csf::PointCloud&, std::vector<int>&, std::vector<int>&)':
../CSFDLL/c2cdist.cpp:11:20: warning: comparison between signed and unsigned int
eger expressions [-Wsign-compare]
[... 124 lines cut ...]
In file included from ../CSFDLL/Cloth.cpp:2:0:
c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\fstream:697:7: note: candidate: std::
basic_ofstream<_CharT, _Traits>::basic_ofstream(const char*, std::ios_base::open
mode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::open
mode = std::_Ios_Openmode]
       basic_ofstream(const char* __s,
       ^
c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\fstream:697:7: note:   no known conve
rsion for argument 1 from 'std::__cxx11::string {aka std::__cxx11::basic_string<
char>}' to 'const char*'
c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\fstream:682:7: note: candidate: std::
basic_ofstream<_CharT, _Traits>::basic_ofstream() [with _CharT = char; _Traits =
 std::char_traits<char>]
       basic_ofstream(): __ostream_type(), _M_filebuf()
       ^
c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\fstream:682:7: note:   candidate expe
cts 0 arguments, 1 provided
c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\fstream:656:11: note: candidate: std:
:basic_ofstream<char>::basic_ofstream(const std::basic_ofstream<char>&)
     class basic_ofstream : public basic_ostream<_CharT,_Traits>
           ^
c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\fstream:656:11: note:   no known conv
ersion for argument 1 from 'std::__cxx11::string {aka std::__cxx11::basic_string
<char>}' to 'const std::basic_ofstream<char>&'
error: command 'C:\\MinGW\\bin\\gcc.exe' failed with exit status 1

jianboqi, 2017-03-14:
Hello,
  It seems that there is no big problem.  std::ofstream can only be constructed with a std::string if you have C++11 or higher. Typically that is done with -std=c++11 (gcc, clang). http://stackoverflow.com/questions/35045781/no-matching-function-for-call-to-stdbasic-ofstreamchar-stdchar-traitscha

So, you can try to add a parameter (-std=c++11) when you are compiling. You can change the setup.py to add this parameter.
csf_module = Extension('_CSF',
                           sources=['csf_wrap.cxx',
                           '../CSFDLL/c2cdist.cpp',
                           '../CSFDLL/Cloth.cpp',
                           '../CSFDLL/CSF.cpp',
                           '../CSFDLL/Particle.cpp',
                           '../CSFDLL/point_cloud.cpp',
                           '../CSFDLL/Rasterization.cpp',
                           '../CSFDLL/XYZReader.cpp'],
                           **extra_link_args=['-std', 'c++11']**
                           )
I do not have MinGW installed, please give a feedback, if it works.

alexguyot, 2017-03-14:
Thanks for helping.

I've added "extra_link_args=['=-std','c++11']" but the result is the same.
I'v also tried with "extra_compile_args=['-std=c++11']" and the build is failing with the following 

running build
running build_py
running build_ext
building '_CSF' extension
C:\MinGW\bin\gcc.exe -mdll -O -Wall -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\
python-2.7.12.amd64\include -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\
..\python-2.7.12.amd64\PC -c csf_wrap.cxx -o build\temp.win-amd64-2.7\Release\cs
f_wrap.o -std=c++11
In file included from c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\random:38:0,
                 from c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\bits\stl_algo.h
:66,
                 from c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\algorithm:62,
                 from csf_wrap.cxx:3640:
c:\mingw\lib\gcc\mingw32\5.3.0\include\c++\cmath:1119:11: error: '::hypot' has n
ot been declared
   using ::hypot;
           ^
error: command 'C:\\MinGW\\bin\\gcc.exe' failed with exit status 1

jianboqi, 2017-03-14:
Hello,
 Add another parameter, it should be OK
csf_module = Extension('_CSF',
                           sources=['csf_wrap.cxx',
                           '../CSFDLL/c2cdist.cpp',
                           '../CSFDLL/Cloth.cpp',
                           '../CSFDLL/CSF.cpp',
                           '../CSFDLL/Particle.cpp',
                           '../CSFDLL/point_cloud.cpp',
                           '../CSFDLL/Rasterization.cpp',
                           '../CSFDLL/XYZReader.cpp'],
                            extra_compile_args=["-fopenmp","-std=c++11","-D_hypot=hypot"],
                     		extra_link_args=["-fopenmp","-std=c++11","-D_hypot=hypot"]
                           )
Also, you should install mingw32-pthreads-win32 library for your mingw. Because parallel computing is very important for this ground filtering algorithm, which will accelerate your program when handling large point cloud file.

alexguyot, 2017-03-14:
Hi,

Thanks for your help. It looks like we're moving forward.
After adding those parameters, the error I'm now getting during build is :

C:\MinGW\bin\gcc.exe -mdll -O -Wall -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\
python-2.7.12.amd64\include -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\
..\python-2.7.12.amd64\PC -c ../CSFDLL/XYZReader.cpp -o build\temp.win-amd64-2.7
\Release\..\csfdll\xyzreader.o -fopenmp -std=c++11 -D_hypot=hypot
../CSFDLL/XYZReader.cpp: In function 'void read_xyz(std::__cxx11::string, csf::P
ointCloud&)':
../CSFDLL/XYZReader.cpp:20:27: error: 'atof' was not declared in this scope
   point.x = atof(x.c_str());
                           ^
error: command 'C:\\MinGW\\bin\\gcc.exe' failed with exit status 1

jianboqi, 2017-03-14:
I don't meet this problem on my computer, may be you need add #include <cstdlib> at the top of the file "XYZReader.cpp". Sorry, I do not have the same environment (WinPython, mingw) installed, so I can not have a complete test. And thank you for testing. we are moving forward.

I also did not use MinGW before, but this code is successfully compiled under Ubuntu with a GCC compiler.

alexguyot, 2017-03-14:
I've added #include < cstdlib> in "XYZReader.cpp".
The build ends with : 

C:\MinGW\bin\gcc.exe -mdll -O -Wall -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\
python-2.7.12.amd64\include -IC:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\
..\python-2.7.12.amd64\PC -c ../CSFDLL/XYZReader.cpp -o build\temp.win-amd64-2.7
\Release\..\csfdll\xyzreader.o -fopenmp -std=c++11 -D_hypot=hypot
writing build\temp.win-amd64-2.7\Release\_CSF.def
C:\MinGW\bin\g++.exe -shared -s build\temp.win-amd64-2.7\Release\csf_wrap.o buil
d\temp.win-amd64-2.7\Release\..\csfdll\c2cdist.o build\temp.win-amd64-2.7\Releas
e\..\csfdll\cloth.o build\temp.win-amd64-2.7\Release\..\csfdll\csf.o build\temp.
win-amd64-2.7\Release\..\csfdll\particle.o build\temp.win-amd64-2.7\Release\..\c
sfdll\point_cloud.o build\temp.win-amd64-2.7\Release\..\csfdll\rasterization.o b
uild\temp.win-amd64-2.7\Release\..\csfdll\xyzreader.o build\temp.win-amd64-2.7\R
elease\_CSF.def -LC:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\..\python-2.
7.12.amd64\libs -LC:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\..\python-2.
7.12.amd64\PCbuild\amd64 -LC:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\..\
python-2.7.12.amd64\PC\VS9.0\amd64 -lpython27 -lmsvcr90 -o build\lib.win-amd64-2
.7\_CSF.pyd -fopenmp -std=c++11 -D_hypot=hypot
C:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\..\python-2.7.12.amd64\libs/li
bpython27.a: error adding symbols: File format not recognized
collect2.exe: error: ld returned 1 exit status
error: command 'C:\\MinGW\\bin\\g++.exe' failed with exit status 1

jianboqi, 2017-03-14:
I just think of a question, your python is 64bit, while your mingw is 32bit. I am not sure whether there will be some problem. It is highly possible they are not compatible with each other. I strongly recommend you using MingGW64. More specifically, you can use TDM-GCC, which has better support of openmp , you can download it http://tdm-gcc.tdragon.net/download. There is a bundle version, which contains MinGW64 and TDM-GCC: http://sourceforge.net/projects/tdm-gcc/files/TDM-GCC%20Installer/tdm64-gcc-5.1.0-2.exe/download

At last, I have a question, you are already using windows platform, why don't you just use visual studio. CSF is developed under visual studio 2013, it will have fewer problem.

alexguyot, 2017-03-15:
Hi jianboqi,

I'm actually using the CSF Python wrapper to use it with laspy and some of my existing python code (I do not know C++).

I've successfully installed TDM-GCC for Win64 (indeed the mingw distrib I used was 32bit).
After adding the -DMS_WIN64 as a compile argument, the build and install run successfully:

However, when calling "import CSF" on the python console, python.exe crashes brutally.
The eventviewer log (sorry it's in french) indicates an issue with the ntdll.dll : 

Nom de l’application défaillante python.exe, version : 0.0.0.0, horodatage : 0x57714567
Nom du module défaillant : ntdll.dll, version : 6.1.7601.23572, horodatage : 0x57fd0651
Code d’exception : 0xc0000374
Décalage d’erreur : 0x00000000000bf3e2
ID du processus défaillant : 0x1cb8
Heure de début de l’application défaillante : 0x01d29d0d91b4da10
**Chemin d’accès de l’application défaillante : C:\WinPython\WinPython-64bit-2.7.12.3Zero\scripts\..\python-2.7.12.amd64\python.exe
Chemin d’accès du module défaillant: C:\Windows\SYSTEM32\ntdll.dll**
ID de rapport : d233f580-0900-11e7-a53a-d6857da1cae0

jianboqi, 2017-03-15:
Hello alexguyot, 
    Actually, I am just in France now, I can understand a little French. It seems that it is close to success. Maybe I will find sometime to try. The WinPython you are using is directly downloaded from the official website? As far as I know, the official binary release of WinPython is built from visual studio, and not from MinGW64. But the binary of CSF is built from MinGW. They are not compatible, this may cause some problems. If you want to do that , you must build a winpython with mingw64 by yourself. https://sourceforge.net/projects/winpython/files/WinPython_Source/Do_It_Yourself/Winpython_2016-01/  I think this is a bit difficult. Do you have some special reasons why you need to use MinGW?  And also, personally, I want to say if you are justing using Python, not C++. Why not try Anaconda, which is crossplatform, and will make life easier. I have use laspy and CSF with Anaconda.

alexguyot, 2017-03-20:
Hi Jianboqi,

Thanks again for your help and time spent for helping.

Yes I'm using the official WinPython setup.
I've also tried using Anaconda. The build and install are throwing this error : 

> ../CSFDLL/Cloth.cpp(383) : error C2664: 'std::basic_ofstream<_Elem,_Traits>::bas
> ic_ofstream(const char *,std::ios_base::openmode,int)' : cannot convert paramete
> r 1 from 'std::string' to 'const char *'
>         with
>         [
>             _Elem=char,
>             _Traits=std::char_traits<char>
>         ]
>         No user-defined-conversion operator available that can perform this conv
> ersion, or the operator cannot be called
> ../CSFDLL/Cloth.cpp(404) : error C2664: 'std::basic_ofstream<_Elem,_Traits>::bas
> ic_ofstream(const char *,std::ios_base::openmode,int)' : cannot convert paramete
> r 1 from 'std::string' to 'const char *'
>         with
>         [
>             _Elem=char,
>             _Traits=std::char_traits<char>
>         ]
>         No user-defined-conversion operator available that can perform this conv
> ersion, or the operator cannot be called
> error: command 'C:\\Users\\AG\\AppData\\Local\\Programs\\Common\\Microsoft\\Visu
> al C++ for Python\\9.0\\VC\\Bin\\amd64\\cl.exe' failed with exit status 2

Then when calling "import CSF" on the python console, python.exe also crashes brutally because of an exception with ntdll.dll (same error with Anaconda and WinPython).

jianboqi, 2017-03-22:
As I have mentioned, the problem is not Anaconda  or winPython. The problem is you are using MinGW to compile CSF, and tries to use it with Python that is compiled with Visual Studio. 

The last solution that I can think of is to install mingw using Anaconda, and using this version of mingw to compile CSF. 
run this command:
`conda install mingw libpython`
After installation, there should be a folder named "MinGW" under Ananconda. As you can see, it also needs a libpython to make it work. I think this is the problem.

If this does not work, I really do not know what happened.  Then the easiest way is to use Visual Studio as the compiler to compile CSF. Visual Studio community version is free now, this is last last solution. I am sorry.

alexguyot, 2017-03-22:
Hi Jianboqui,

I've installed mingw and libpython from Anaconda, and uninstalled TDM-GCC.
The "python setup.py install" of CSF is now complaining about "omp.h" ...

running install
running build
running build_py
running build_ext
building '_CSF' extension
C:\ProgramData\Anaconda2\Scripts\gcc.bat -DMS_WIN64 -mdll -O -Wall -IC:\ProgramD
ata\Anaconda2\include -IC:\ProgramData\Anaconda2\PC -c csf_wrap.cxx -o build\tem
p.win-amd64-2.7\Release\csf_wrap.o
C:\ProgramData\Anaconda2\Scripts\gcc.bat -DMS_WIN64 -mdll -O -Wall -IC:\ProgramD
ata\Anaconda2\include -IC:\ProgramData\Anaconda2\PC -c ../CSFDLL/c2cdist.cpp -o
build\temp.win-amd64-2.7\Release\..\csfdll\c2cdist.o
In file included from ../CSFDLL/c2cdist.h:5:0,
                 from ../CSFDLL/c2cdist.cpp:1:
../CSFDLL/Cloth.h:32:17: fatal error: omp.h: No such file or directory
compilation terminated.
error: command 'C:\\ProgramData\\Anaconda2\\Scripts\\gcc.bat' failed with exit s
tatus 1

Well, I think I'm going to give up.
I might give it a try with Visual Studio later, but since I don't know C++, I believe it will not be straight forward ;-)

For non-developer like me, do you plan to release the "CSFDemo.exe" with las/laz support ?

Thanks a lot,
Alex

jianboqi, 2017-03-22:
Hello Alex,
   before using Visual Studio, try this:
https://github.com/jianboqi/CSF/tree/master/python/PrecompiledForAnaconda2

I just uploaded a precompiled version of CSF for Anaconda2 (64bit), put all the *CSF*.* files into Anaconda2/lib/site-packages.  It maybe work.

For the CSFDemo.exe it is based on C++, if I want to support las/laz, I should use liblas. But I think it will not have many users, because if you want to embed it into a processing chain. it is not convenient to read las , and do filtering, and then write las, then read by another program in this processing chain, especially for very large files.  

At last, if this does not work, you should try visual studio, I am sure that it will work. because we have compile with visual studio on many computers.

alexguyot, 2017-03-23:
Hi Jianboqui !

Good news ! Your pre-compiled version of CSF works on Win64/Anaconda and on Win64/WinPython environnements ! I've been able to use it with laspy and eventually generate some filtered LAS files. Thanks a lot.

I'm just facing a stange issue with the cfs=CSF.CSF() which raise a "TypeError: 'module' object is not callable" when used within a script (your script exemple) but works fine when called from the python command prompt. I tried some different import (from CSF import CSF, from CSF import *) but without any luck.

Do you have any idea ?

Last question, do you have some explanations (and exemples) of the advanced CSF parameters (such as rigidness) ?

Thanks a lot,
Alex

jianboqi, 2017-03-23:
Hello, Alex,
   How your scripts is written, I tested on my computer and I do not meet this kind of problem. And how do you run the script? from cmd window：`python a.py`, something like this.  Or do you use an integrated development software, such as pycharm. If so, have you config a right python version for the software. All these may cause problem.  On my computer, both of these methods are working properly. So you must pay attention to them.

As for the rigidness, it represents the properties of the cloth. When rigidness is smaller (e.g. =1), it is very soft. When it is large value, it is hard.  This effect can be illustrated by the following figure.

[image]

alexguyot, 2017-03-23:
Hi,

Thanks for the rigidness explanation.

My script is a copy/paste of your example (see below) and I call it on the anaconda command prompt with 'python testCFS.py' (I've also tried with spyder/anaconda and got the same message.

Strangely enough, if copy/paste all lines directly on the anaconda python prompt, it works fine.

<
# coding: utf-8
import laspy
import CSF
import numpy as np

csf=CSF.CSF()

inFile = laspy.file.File(r"D:\input.laz", mode='r') # read a las file
points = inFile.points
xyz = np.vstack((inFile.x, inFile.y, inFile.z)).transpose() # extract x, y, z and put into a list

# prameter settings
csf.params.bSloopSmooth = False
csf.params.cloth_resolution = 2
# more details about parameter: http://ramm.bnu.edu.cn/projects/CSF/download/

csf.setPointCloud(xyz)
ground = CSF.VecInt()  # a list to indicate the index of ground points after calculation
non_ground = CSF.VecInt() # a list to indicate the index of non-ground points after calculation
csf.do_filtering(ground, non_ground) # do actual filtering.

outFile = laspy.file.File(r"D:\ground.las", mode='w', header=inFile.header)
outFile.points = points[ground] # extract ground points, and save it to a las file.
outFile.close() # do not forget this
inFile.close()
>

closed 2018-07-19 by jianboqi