jianboqi/CSF #3: Missing CSF.i swig interface file

pandu-rao, 2017-07-24:
Hello,

I don't see a swig interface file in the repository.

Could you please add that?

It will help us make bug-fixes, etc.

Thanks.

jianboqi, 2017-07-25:
Hello,
    We did not add this file because it is usually not used by an end-user.
    If you need it, I will be very happy to share it. 
   I have pushed it to the repository.

pandu-rao, 2017-07-25:
Thank you.

While waiting for the file, here's what I came up with:
```
%module CSF
%{
  #define SWIG_FILE_WITH_INIT
  #include "../CSFDLL/CSF.h"
%}

%include "std_vector.i"

namespace std
{
    %template(VecInt) vector<int>;
    %template(Vecf) vector<float>;
    %template(VecVecf) vector< vector<float> >;
}

%include "../CSFDLL/CSF.h"
```

As you can see, this line is missing from my version:
`%include "std_string.i"`

I will use your version.

One question i have is on these two lines:
```
    %template(Vecf) vector<float>;
    %template(VecVecf) vector< vector<float> >;
```

Can I use these instead?
```
    %template(VecFloat) vector<float>;
    %template(VecVecFloat) vector< vector<float> >;
```
Thanks,
Pandu

jianboqi, 2017-07-26:
I think you can use that, because it is just a vector name defined by myself.

closed 2017-07-27 by jianboqi