jianboqi/CSF #6: why use gravity like drop process?

riyueshi, 2017-08-16:
Hi, jianboqi
  I am curious what is the reason CSF using gravity like drop process? In order to hit the ground faster? Or is there any other reason using this model?
 Miller

jianboqi, 2017-08-21:
Hi Miller,
    We use gravity just to simulate the process that cloth drops to the ground, which is more natural and easier to understand by users. Because what we want to do is to fit the cloth with the shape of the terrain, this process just looks like a cloth drops to the ground (inverted).

closed 2017-08-29 by jianboqi

riyueshi, 2017-08-30:
Thanks, but I think using gravity to compute the step dosen't have any help for user to understand. And this may result in very big drop step if the points contain outlier.