jianboqi/CSF #8: Is it possible to set thresholds for different Las files?

pradAI, 2018-07-24:
Hi, I have set threshold values for one LAS file like below but need to change thresholds for neighbor regions. Is it possibly to set thresholds automatically to separate ground and non ground regions.

LasFIle-1

csf.params.bSloopSmooth = True
csf.params.cloth_resolution = 0.5
csf.params.rigidness = 5
csf.params.time_step = 0.65;
csf.params.class_threshold = 0.02
csf.params.interations = 500;

Las File-2
csf.params.bSloopSmooth = True
csf.params.cloth_resolution = 0.25
csf.params.rigidness = 5
csf.params.time_step = 0.65;
csf.params.class_threshold = 0.02
csf.params.interations = 500;

jianboqi, 2018-07-25:
Hello,
   I do not understand your question well, CSF works on the whole point cloud, and it is difficult to use different parameters in different regions of the same cloud. If you want to use different parameters for different las file, why not do it in seperate process. I don't know if I well understand your question.
Jianbo

pradAI, 2018-07-25:
Thanks for the suggestion.

closed 2018-08-01 by jianboqi

pradAI, 2018-12-18:
How to choose automatic threshold values for different regions of point cloud data ?