jianboqi/CSF #11: function CSF::setPointCloud get wrong index from points

colinlin1982, 2019-01-03:
Hi. I think source code in src/CSF.cpp
`void CSF::setPointCloud(double *points, int rows) {
    #define A(i, j)    points[i + j * rows]

    for (int i = 0; i < rows; i++) {
        csf::Point p;
        p.x = A(i, 0);
        p.y = -A(i, 2);
        p.z = A(i, 1);
        point_cloud.push_back(p);
    }
}`
A(i, j) is wrong defined. for rows = 3,  i from 0 to 3, x, y, z index will be:
p.x = A(0, 0) = points[0 + 0 * 3] = points[0]
p.y = -A(0, 2) = -points[0 + 2 * 3] = -points[6]
p.z = A(0, 1) = points[0 + 1 * 3] = points[3]
p.x = A(1, 0) = points[1 + 0 * 3] = points[1]
p.y = -A(1, 2) = -points[1 + 2 * 3] = -points[7]
p.z = A(1, 1) = points[1 + 1 * 3] = points[4]
p.x = A(2, 0) = points[2 + 0 * 3] = points[2]
p.y = -A(2, 2) = -points[2 + 2 * 3] = -points[8]
p.z = A(2, 1) = points[2 + 1 * 3] = points[3]

That means input points contains data in an odd order:
x0, x1, x2, y0, y1, y2, z0, z1, z2

but if the macro function defined like this:
#define A(i, j)    points[i * rows + j]

Everything will be OK.

So is it a bug or I wrongly understand the function?

colinlin1982, 2019-01-03:
It's my honor.

Jean-Romain, 2019-01-03:
What are the consequences of such bug. Do I need to fixed it as soon as possible in RCSF?

jianboqi, 2019-01-03:
the function which has been changed here is seldom used: void CSF::setPointCloud(double *points, int rows) . To be honest, I never used it before, that why I did not have problem. But anyway, i am very happy to makes fewer bugs in the code. (*^_^*)

jianboqi, 2019-01-03:
This function read a one-dimension array as point cloud into CSF.

closed 2019-01-04 by colinlin1982