jianboqi/CSF #20: When i use the c++ version, I got a bug.

WAN96, 2019-09-06:
I just implement this program into removing the ground lidar point. And the lidar was equipped on a car. I use kitti dataset to test this program. But i got a bug.
 [0] Configuring terrain...
[0] Configuring cloth...
[0]  - width: -2147483644 height: -2147483644
[0] Rasterizing...
[0] Simulating...
Segmentation fault
the parameter that i choose was:
  csf.setPointCloud(cld);
  csf.params.bSloopSmooth = false;
  csf.params.class_threshold = 0.12;
  csf.params.cloth_resolution = 1;
  csf.params.interations = 200;
  csf.params.rigidness = 1;

jianboqi, 2019-09-07:
what the cld is? Maybe the point cloud is not successfully loaded, because I noticed that - width: -2147483644 height: -2147483644, which is not normal. Please show me your complete code. Thank you

WAN96, 2019-09-19:
> what the cld is? Maybe the point cloud is not successfully loaded, because I noticed that - width: -2147483644 height: -2147483644, which is not normal. Please show me your complete code. Thank you

I am sorry  for the late reply, it looks like I solved the problem.

closed 2021-01-03 by jianboqi