jianboqi/CSF #21: CSF is non-deterministic when multithreading.

dbrookes96, 2019-09-10:
The CSF ground filter does not give consistent results when multithreading. i.e. between runs we get slightly different ground and non-ground classifications.

I've attached a CSV point cloud I used for testing that contains 500,000 points ([trimble_500k.zip](https://github.com/jianboqi/CSF/files/3593166/trimble_500k.zip)). For testing I'm running the CSF filter with the settings:
```
Scene = Relief
Slope processing = True
Cloth Resolution = 1.0
Max Iterations = 500
Classification Threshold = 0.5
Export Cloth Mesh = False
```
Out of 5 runs I got the following number of ground points.
```
1. 71,275
2. 71,276
3. 71,275
4. 71,275
5. 71,276
```

Setting the environment variable OMP_NUM_THREADS to 1 results in each run returning 71,274 ground points. I'd assume that this would be the correct result?

As a workaround we removed all `#pragma omp parallel for` statements from CSF (there is only a few). This resulted in seemingly consistent results between runs. I also didn't observe too much of a drop in performance which was surprising, even for a ~63 million point cloud.

dbrookes96, 2019-09-10:
I've also logged this with CloudCompare since they use a modified version of this library (https://github.com/CloudCompare/CloudCompare/issues/953).

jianboqi, 2019-09-16:
Hello,
  Thank you for report this problem.
 This can happen when omp parallel is enabled, because the sequence of parallel computing cannot be controlled, which causes the final shape of cloth to be different, thus the classification resutls will be different. However, this will not cause a serious problem. 
   The number of the points is not the major factor that influences the computing time, the extent of the point cloud  and the cloth resolution are the most important factors.

closed 2021-01-03 by jianboqi