jianboqi/CSF #27: Is is possible to get the cloth

Jean-Romain, 2019-12-17:
Hi, I'm wondering if it is possible to return the cloth i.e. return the coordinates of the nodes programmatically. There is a an option `exportCloth` in `CSF::do_filtering` but it seems to write something in file. I'd like to get a C++ object I could transform into a raster in R. I can modify the code if needed.
Thanks.

yulongyu, 2019-12-18:
when you dont give the params,exportCloth will equal false,you can change the code ,let it alway equal true

Jean-Romain, 2019-12-18:
@yulongyu I don't want to write the cloth in file. I want to get it in a C++ object at runtime.

yulongyu, 2019-12-18:
it is a vector<point> xyz , i dont know what is you meaning, you what use it in R language?

Jean-Romain, 2019-12-18:
https://github.com/Jean-Romain/RCSF

yulongyu, 2019-12-19:
in csf.h , std::vector<Particle> particles //it saved the cloth

yulongyu, 2019-12-19:
i know what you want to do ...... but i think it is not recommend to return a lot points to another language as parameter，In a nutshell, your question is how to pass complex types to another programming language

closed 2021-01-03 by jianboqi