jianboqi/CSF #34: 关于ISPRS数据集复现出现的一些问题

laiming997, 2020-12-31:
使用开源的代码和论文所说的参数对ISPRS数据集进行测试，已提前对ISPRS的原始点云进行了统计学滤波过滤低点了，但是一类误差和论文所述差距很大，请问会是什么原因导致的呢

jianboqi, 2021-01-03:
原始版本的程序再github开源之后，为了提高计算效率，减少库依赖，很多方面都有过修改了，现在不需要任何外部库。修改之后没有再用ISPRS数据进行测试测试，如果要复现文章结果的话，可以使用最早的原始代码。已经通过邮箱发给你了，你可以试试。

closed 2021-01-03 by jianboqi

laiming997, 2021-01-05:
非常感谢您提供的源码，我对比了下与现在github上的区别，其中在raster和赋予内部作用力，即rigidness有区别。我对您提供的源码进行了编译，因为提供的lib是release的无法debug，跑出来的效果更差了无法查看什么原因，请问下您这份代码是当时跑的那份嘛，非常感谢

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/jianboqi/CSF/issues/34#issuecomment-753558678",
"url": "https://github.com/jianboqi/CSF/issues/34#issuecomment-753558678",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

jianboqi, 2021-01-05:
最原始的代码找不到了，不过给你的这个代码是根据之前部分代码恢复的。我这边之前有人测试过，这个结果还是比较接近文章里的，这个就是给你的那个代码编译的，你可以试一下。

[test.zip](https://github.com/jianboqi/CSF/files/5770810/test.zip)

laiming997, 2021-01-05:
您好，你给我的压缩包了是个可执行文件，您说这个可执行文件时您前几天发给我的CSF-原始源码.zip压缩包编译后的吗？我执行了一下这次压缩包里的可执行文件，发现Rasterizing的时候有raster1、raster2、raster3，是不是对地形根据起伏进行了3个
量级的分区，这部分代码在之前的压缩包里好像没有体现。我的问题比较多，因为最近毕业论文基于你们工作之下的，时间比较着急，所以一直打扰到您，非常抱歉，也非常感谢您的耐心指导。

https://github.com/jianboqi/CSF/issues/34#issuecomment-754682065

jianboqi, 2021-01-11:
我们这边测试了，没有发现什么raster1，raster2之类的，你可以直接发邮件吧，沟通更方便一些，发我们一个截图jianboqi@126.com 看看是什么问题。