jianboqi/CSF #36: D8021 invalid numeric argument '/Werror' CSF

FavorMylikes, 2021-08-28:
Is there one have seen this error before when build solution?

jianboqi, 2021-08-31:
Hi,
I just noticed that there is a problem when building on Windows, you can remove /Werror option:
    target_compile_options(CSF PRIVATE -Werror -Wall -Wextra)
When building on windows, you can remove this line in src/CMakeLists.txt

Crghilardi, 2021-08-31:
@FavorMylikes Would you be able to test the fix in [this commit](https://github.com/jianboqi/CSF/pull/38/files) and see if that fixes it? I do not have access to a Windows machine currently

FavorMylikes, 2021-09-01:
@Crghilardi Seems ok, I replace `/Werror to /W4`, and remove  `/Wextra` manually.

I use cmake to build vs solution, and not find where the code is.

Here is what I changed at `CSF/CMakeList.txt`.

```cmake
if(OpenMP_CXX_FOUND)
    target_link_libraries(CSF PUBLIC OpenMP::OpenMP_CXX)
    target_compile_options(CSF PRIVATE -W4 -Wall)
endif()
```

PR #38 (2022-11-19): add windows/linux branch to compiler options
This pull request adds a `if/else` branch for compiler options. From https://cmake.org/cmake/help/latest/command/add_compile_options.html#example

should fix #36 ?
closed 2022-11-19 by jianboqi