jianboqi/CSF #39: install TARGETS given no ARCHIVE DESTINATION for static library target CSF

TheAdamP, 2021-09-06:
Hey might be a stupid question but trying to build on Linux gives me this error:
CMake Error at src/CMakeLists.txt:31 (install):
  install TARGETS given no ARCHIVE DESTINATION for static library target "CSF".

Have I missed an obvious step in the installation? I just downloaded file made a build directory then tried to cmake.

Crghilardi, 2021-09-07:
Need some more info, what was the exact `cmake` command you ran?

TheAdamP, 2021-09-07:
So I put the master folder into my build file and followed the install guide. Specifically I ran `cmake CSF-master` and `cmake CSF-master/src` whilst inside the build folder

Crghilardi, 2021-09-07:
It sounds like you downloaded the zip file of the repo instead of `git clone` command? Either way it should work fine.
I am on `Ubuntu` and just pulled down a copy. Here is what works for me with some notes added. Try these

```
#download and unzipped from https://github.com/jianboqi/CSF/archive/refs/heads/master.zip
#show what's in the top level directory
~/Downloads$ cd CSF-master/ && ls
CMakeLists.txt  CSFDemo  LICENSE.txt  matlab  python  README.md  src

#let's do an out-of-tree build to keep build files separate
~/Downloads/CSF-master$ mkdir build
~/Downloads/CSF-master$ cd build

#call cmake with default settings, note the double period since we are in CSF-master/build and the CMake files are in CSF-master
~/Downloads/CSF-master/build$ cmake ..
-- The CXX compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Configuring done
-- Generating done
-- Build files have been written to: /Downloads/CSF-master/build

#compile everything, default setting is making a static library
~/Downloads/CSF-master/build$ make -j4
Scanning dependencies of target CSF
[ 44%] Building CXX object src/CMakeFiles/CSF.dir/c2cdist.cpp.o
[ 44%] Building CXX object src/CMakeFiles/CSF.dir/Constraint.cpp.o
[ 44%] Building CXX object src/CMakeFiles/CSF.dir/Cloth.cpp.o
[ 44%] Building CXX object src/CMakeFiles/CSF.dir/CSF.cpp.o
[ 55%] Building CXX object src/CMakeFiles/CSF.dir/Particle.cpp.o
[ 66%] Building CXX object src/CMakeFiles/CSF.dir/point_cloud.cpp.o
[ 77%] Building CXX object src/CMakeFiles/CSF.dir/Rasterization.cpp.o
[ 88%] Building CXX object src/CMakeFiles/CSF.dir/XYZReader.cpp.o
[100%] Linking CXX static library libCSF.a
[100%] Built target CSF

```

TheAdamP, 2021-09-07:
Hi so I tried this exactly as followed and got the similar error:
adam@adam-Blade-Stealth:~/Downloads$ cd CSF-master
adam@adam-Blade-Stealth:~/Downloads/CSF-master$ mkdir build
adam@adam-Blade-Stealth:~/Downloads/CSF-master$ cd build
adam@adam-Blade-Stealth:~/Downloads/CSF-master/build$ cmake ..
-- The CXX compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
CMake Error at src/CMakeLists.txt:31 (install):
  install TARGETS given no ARCHIVE DESTINATION for static library target
  "CSF".

-- Configuring incomplete, errors occurred!
See also "/home/adam/Downloads/CSF-master/build/CMakeFiles/CMakeOutput.log".

I followed the steps exactly so a little unsure of whats going on, sorry I am fairly new to cmake in general

Crghilardi, 2021-09-07:
That is strange behavior. As a quick workaround, you should be able to comment that line out in the `src/CMakeLists.txt` like

```
install(TARGETS CSF LIBRARY DESTINATION lib)
#install(TARGETS CSF ARCHIVE DESTINATION lib)
install(FILES ${CSF_HEADERS} DESTINATION include)
```

What version of cmake are you using? (`cmake --version`), I am testing on my side with `3.15.5`

TheAdamP, 2021-09-09:
Commenting this out still gave the same error unfortunately. However commenting the line above fixed this issue, unsure if this will have an effect later on. I am running cmake `3.10.2-1`

chaohe1998, 2022-05-13:
> 
Hello, have you solved this problem

shuqiu0202, 2022-07-26:
install(TARGETS CSF
    LIBRARY DESTINATION lib
    ARCHIVE DESTINATION lib
)
install(FILES ${CSF_HEADERS} DESTINATION include)

I have same problem in ubuntu 18.04, cmake 3.10.2, gcc 7.5, change old to this can work.

silent-zhou, 2023-03-27:
updating the  version of cmake will solve your problem