jianboqi/CSF #41: Error Installing CSF

ceavilest, 2021-09-13:
Good Morning I was having the following issues trying to install CSF in python 3.8 I get the following error
CSF_wrap.cxx(3135): fatal error C1083: Cannot open include file: '../src/CSF.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Enterprise\\VC\\Tools\\MSVC\\14.24.28314\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

I do have installed Visual Studio.

Ironayron, 2021-11-24:
Got the same error, is there any workaround?

jianboqi, 2022-05-01:
Hi, I tested on my PC with Python 3.7, python 3.8 and Python 3.9. All is OK, currently, I am not sure what's the problem. I will check it later with a new PC to see if I can reproduce this problem. How do you run your code, are you in the python folder under the CSF directory?

Ironayron, 2022-05-01:
Hey Jianboqi,

i was at the end as well able to ran the code!!
Thank you very much, the CSF filtering is a very usefull tool

Von Outlook<http://aka.ms/weboutlook> gesendet.

xuxiuzhi2627, 2022-08-26:
> Hi, I tested on my PC with Python 3.7, python 3.8 and Python 3.9. All is OK, currently, I am not sure what's the problem. I will check it later with a new PC to see if I can reproduce this problem. How do you run your code, are you in the python folder under the CSF directory?

Can this algorithm be used in python 2.7?