jianboqi/CSF #43: clang: error: unsupported option '-fopenmp' on Apple MacOS Monterey

dluks, 2022-04-30:
Device info:
- Macbook Pro, mid-2015, Intel processor
- macOS Montery (12.3)

After `git clone`ing and `cd`ing into the `python` directory, running `python setup.py build` results in the following error:
```bash
$ python setup.py build
running build
running build_py
running build_ext
building '_CSF' extension
clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/lusk/miniconda3/envs/PointClouds/include -fPIC -O2 -isystem /Users/lusk/miniconda3/envs/PointClouds/include -I/usr/local/opt/llvm/include -I/Users/lusk/miniconda3/envs/PointClouds/include/python3.9 -c ../src/CSF.cpp -o build/temp.macosx-10.9-x86_64-3.9/../src/CSF.o -fopenmp
clang: error: unsupported option '-fopenmp'
error: command '/usr/bin/clang' failed with exit code 1
```
I've tried installing the homebrew gcc package and then linking gcc to use gcc-11, but the issue persists.

jianboqi, 2022-05-01:
Hi, this may be the problem of mac configuration. I never tested it with MAC. I do a simple google, maybe this is caused by the compiler clang that is from XCode. you can try to install a gcc by yourself, such as brew install gcc

dluks, 2022-05-02:
Thanks @jianboqi, but unfortunately I've already tried this. Here's what I have tried, specifically:

1. `brew install gcc`
2. `export PATH=/usr/local/bin:$PATH`
3. `ln -s gcc-11 gcc`
4. 'gcc -v` to confirm that I am using the new symlinked gcc

Unfortunately this still results in the same error.

miniygq, 2022-05-20:
遇到同样问题