jianboqi/CSF #44: New feature wrapped in laspy example returns Attribute Error

UrAvgProgrammer, 2022-06-27:
```
# coding: utf-8
import laspy
import CSF
import numpy as np

inFile = laspy.read("D:\\labeled_dateset\\aoi_moon_park.las")

xyz = np.vstack((inFile.x, inFile.y, inFile.z)).transpose() # extract x, y, z and put into a list

csf = CSF.CSF()

# prameter settings
csf.params.bSloopSmooth = False
csf.params.cloth_resolution = 0.5
# more details about parameter: http://ramm.bnu.edu.cn/projects/CSF/download/

csf.setPointCloud(xyz)
ground = CSF.VecInt() # a list to indicate the index of ground points after calculation
non_ground = CSF.VecInt() # a list to indicate the index of non-ground points after calculation
csf.do_filtering(ground, non_ground) # do actual filtering.

outFile = laspy.create(point_format=inFile.header.point_format, file_version=inFile.header.version)
outFile.points = inFile.points[ground] # extract ground points, and save it to a las file.
outFile.write(r"out.las")
```

Output/Traceback:
```
Traceback (most recent call last):
  File "test.py", line 26, in <module>
    outFile.points = inFile.points[ground] # extract ground points, and save it to a las file.
  File "C:\Anaconda\envs\env\lib\site-packages\laspy\lasdata.py", line 367, in __setattr__
    super().__setattr__(key, value)
  File "C:\Anaconda\envs\env\lib\site-packages\laspy\lasdata.py", line 98, in points
    if new_points.point_format != self.point_format:
AttributeError: 'numpy.ndarray' object has no attribute 'point_format'
```

closed 2022-06-27 by UrAvgProgrammer

massijjeong, 2023-01-30:
how did you solve this problem?

weypro, 2023-03-20:
My solution:
```
# or points[ground]
outFile.points = laspy.point.record.PackedPointRecord(points[non_ground],laspy.PointFormat(6))
```
@jianboqi If it is convenient for you, you can modify the example or resolve the conflict fundamentally.