jianboqi/CSF #49: I have a confusion regarding whether y is the height value in the point cloud and if so, should I switch y and z column in my point cloud data?

Heavy0strich, 2023-04-08:
in _**CSF/src/Cloth.cpp**_ inside function `Cloth::timeStep()` on lines **111- 118** where we are calculating the _**diff**_, it assigns the absolute difference in the **_y_** values of the particle's **old position** and **new position**. As mentioned in the paper the particles are constrained to move in the vertical direction. The point cloud data that I am using has height values in _**z**_, so should I switch the _**y**_ and _**z**_ columns in my data?

jianboqi, 2023-04-09:
In CSF, the internal coordinate is a bit different from our common coordinate (e.g., z in vertical direction). In CSF,  y is in vertical direction. When we read point by using read_xyz (in XYZReader.cpp), we already switch the y and z, and inverted the point cloud upside down. If you do not use this function, you should switch it by yourself.
```
csf::Point point;
        point.x = atof(x.c_str());
        point.y = -atof(z.c_str());
        point.z = atof(y.c_str());
```

Heavy0strich, 2023-04-10:
Thanks for the quick response, I will update my code accordingly

closed 2023-04-10 by Heavy0strich