jianboqi/CSF #60: do_filtering slower in CPP than Python

thebigpotatoe, 2025-01-31:
Hi, 

Thanks for such an amazing library, its running really well over my datasets to segment ground out,

I am however running into some very weird behavior where the library runs faster in python over a similar implementation in cpp.

For example on my machine, the python code i have runs in approx 60ms per pointcloud whereas the same routine using the same dataset in cpp runs in approx 300ms per pointcloud.

Is this to be expected or has this been seen before at all? I've attempted to configure openmp and other compile time optimisations to no avail and I've also deep dived into profiling  both implementations and haven't found anything that suggests why it should be running slower.

jianboqi, 2025-07-08:
This might because of the different compiler? I will test it.

cama36, 2025-07-08:
Any results? I got the same results.
— with results from CloudCompare being both faster and more accurate.

thebigpotatoe, 2025-08-13:
I don't think so? 

All of the code I have written myself for Linux exclusively, just wrapping your library. Not using the demo script.

Is there possibly a better memory management technique that the demo uses?